Screenly/Anthias #2406: [BUG] White screen of death

mgroves, 2025-07-19:
### Description

Intermittantly, the display goes white, and stays there. Reboot seems to correct it, but it eventually goes white again.

### Steps to Reproduce

I don't really have any steps, but I can tell you that the media I've tried is a single URL or a single image, alone. It goes white for each.

I'm happy to post logs, if you tell me which would be most helpful.

### Expected Behavior

Not going white: keep displaying content.

### Screenshots

n/a, imagine a white screen.

### Environment

- Device: Raspberry Pi 5 Model B Rev 1.0
- OS: Debian GNU/Linux 12 (bookworm)
- Anthias Version: master@2973d59
- Installation Method: curl

nicomiguelino, 2025-07-19:
@mgroves, while I'm taking a look at it, I would like to ask for more details:
* Did you install Anthias on top of Raspberry Pi OS Lite?
* Would you be able to provide at least the logs of the `anthias-server` and the `anthias-viewer` services? Please use [this documentation](https://github.com/Screenly/Anthias/tree/master/docs#how-to-get-logs-from-anthias) as a guide.

mgroves, 2025-07-19:
> Did you install Anthias on top of Raspberry Pi OS Lite?

I don't think it's Lite. Here's the output of `cat /etc/os-release`

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

> Would you be able to provide at least the logs of the anthias-server and the anthias-viewer services? Please use [this documentation](https://github.com/Screenly/Anthias/tree/master/docs#how-to-get-logs-from-anthias) as a guide.

I've attached the output of `docker compose logs -f anthias-server > server.logs` and `docker compose logs -f anthias-viewer > viewer.logs`

[viewer.logs.txt](https://github.com/user-attachments/files/21329750/viewer.logs.txt)
[server.logs.txt](https://github.com/user-attachments/files/21329749/server.logs.txt)

> Is your display connected to the HDMI0 port (the one that's close to the USB-C port) or to the other HDMI (HDMI1) port?

I think it's HDMI0. I'm not physically with the device right now, but ChatGPT suggested `for f in /sys/class/drm/*/status; do echo "$f: $(cat $f)"; done`, and here's the output:

```
redacted@raspberrypi:~ $ for f in /sys/class/drm/*/status; do echo "$f: $(cat $f)"; done
/sys/class/drm/card1-HDMI-A-1/status: connected
/sys/class/drm/card1-HDMI-A-2/status: disconnected
/sys/class/drm/card1-Writeback-1/status: unknown
/sys/class/drm/card1-Writeback-2/status: unknown
```

nicomiguelino, 2025-07-19:
@mgroves, thanks for providing me the log files and other details.

What do you get when you run the following?

```bash
ls /dev/fb*
ls /dev/dri
```

I tried doing a clean install and an upgrade, but I was not able to reproduce your issue.
It might have something to do with the monitors that we are using, but let's try to continue ruling things out.

mgroves, 2025-07-19:
I should mention also that I've been running Anthias since March 2025, and it's only been happening recently.

nicomiguelino, 2025-07-19:
Thanks. I'll continue to take a look at it.

nicomiguelino, 2025-08-25:
@mgroves, are you still having issues? I checked the logs that you provided and saw nothing unusual.

Would you be able to provide the specs (e.g., resolution) of the monitor used?