Screenly/Anthias #2484: [BUG] Anthias default startup screen appears between every asset

hoytler, 2025-08-30:
### Description

Note: I just started using Anthias and immediately ran into this bug.

On a fresh install of Anthias on RPiOS Lite, using the default install method to deploy docker containers:  after adding assets (no defaults used), the Anthias "To manage the content on this screen..." page appears in between each asset refresh for about 1 second. This bug is seemingly persistent and occurs after I completely wipe all containers from the system and reinstall from scratch.

### Steps to reproduce the behavior:

1. install Anthias on RPiOS Lite per the instructions
2. bash <(curl -sL https://install-anthias.srly.io)
3. Select no network management (using WIFI), No full-upgrade, reboot after install.
4. Do not use default assets, load 4 assets (each with 1 minute refresh)
5. Anthias "To manage the content on this screen..." page begins appearing in-between each asset load for about 1 second. 

Note: all my assets do load properly.

### Expected Behavior

I shouldn't be seeing the initial setup screen after adding at least one asset.

### Environment

Load Average	0.19
Free Space	47G
Memory	
Total: 3679 MiB
Used: 612 MiB
Free: 2255 MiB
Shared: 66 MiB
Buff: 58 MiB
Available: 2931 MiB
Display Power (CEC)	None
Device Model	Raspberry Pi 4 Model B Rev 1.1
Anthias Version	[master@276f06c](https://github.com/Screenly/Anthias/commit/276f06c)

- Installation Method: [bash script]

nicomiguelino, 2025-08-31:
@hoytler, let me try to replicate this issue. I tried updating Anthias (not fresh install) and haven't got any issues. I'm trying a clean install now.

hoytler, 2025-08-31:
- These are just local webpages, served up a another host on the local network. These pages contain some image assets, but no video.

There is nothing in the viewer logs that stay from the output shown below:
`
anthias-viewer-1  | ./bin/start_viewer.sh: line 39: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
anthias-viewer-1  | Loading browser...
anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/standby.png
anthias-viewer-1  | Generating asset-list...
anthias-viewer-1  | Current url is http://anthias-nginx:80/splash-page
anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/standby.png
anthias-viewer-1  | Showing asset Weather (webpage)
anthias-viewer-1  | Current url is http://somehost.local:8080/weather/
anthias-viewer-1  | Sleeping for 185
anthias-viewer-1  | Showing asset Wildfire (webpage)
anthias-viewer-1  | Current url is http://somehost.local:8080/fire/
anthias-viewer-1  | Sleeping for 65
anthias-viewer-1  | Showing asset Traffic (webpage)
anthias-viewer-1  | Current url is http://somehost.local:8080/traffic/
anthias-viewer-1  | Sleeping for 65
anthias-viewer-1  | Showing asset Earthquake Map (webpage)
anthias-viewer-1  | Current url is http://somehost.local:8080/usgs/
anthias-viewer-1  | Sleeping for 65
anthias-viewer-1  | Showing asset Weather (webpage)
anthias-viewer-1  | Current url is http://somehost.local:8080/weather/
anthias-viewer-1  | Sleeping for 185
...
`

closed 2025-09-01 by hoytler