Screenly/Anthias #2493: [FEATURE] Replace apt-key usage

romainreignier, 2025-09-05:
### Description

I have tried to run the install script on Debian 13 and the Docker key installation step in the ansible playbook fail because `apt-key` is no longer available.

I know that Debian 13 is not supported yet but it is quite difficult nowadays to find the Debian 12 iso and even the link in your docs points to Debian 13 download page. Other users might encounter the issue. 

### Solution

`apt-key` has been deprecated for some years now.
See the Docker instructions here that do not use `apt-key` anymore: https://docs.docker.com/engine/install/debian/#install-using-the-repository

DongerZone, 2025-10-03:
Are there any alternative instructions that can be used in the meanwhile?

romainreignier, 2025-10-06:
Now that RaspberryPi OS is based on Debian 13, this issue will need to be fixed: https://www.raspberrypi.com/news/trixie-the-new-version-of-raspberry-pi-os/