Screenly/Anthias #2520: White Screen On All Anthias Installs - HELP

TRS4242, 2025-09-30:
Definitely a novice user to both Anthias and PiOS here, but here is my issue.

After upgrading the 5 Anthias displays around campus to the latest version, all show a white screen upon boot and nothing else.  All units are Pi 4s running Pi OS Lite (64bit) Bookworm. Thinking that something in my configuration might have messed something up, I installed Anthias on a fresh install of the same OS with the same results.  I'll be glad to upload any logs or other info, just be sure to give me instructions on were to access them - again, a Pi OS/Linux noob here!

me-abhishekpal, 2025-09-30:
More info 

~$ docker run -t  screenly/anthias-viewer:latest-x86
./bin/start_viewer.sh: line 39: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
Config-file /root/.screenly/screenly.conf missing. Using defaults.
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 148, in _get_module_details
  File "<frozen runpy>", line 112, in _get_module_details
  File "/usr/src/app/viewer/__init__.py", line 20, in <module>
    from settings import LISTEN, ZmqConsumer, settings
  File "/usr/src/app/settings.py", line 160, in <module>
    settings = AnthiasSettings()
               ^^^^^^^^^^^^^^^^^
  File "/usr/src/app/settings.py", line 84, in __init__
    self.save()
  File "/usr/src/app/settings.py", line 143, in save
    with open(self.conf_file, "w") as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/root/.screenly/screenly.conf'

DerCookieLiker, 2025-09-30:
Also my view is crashing with the same error @me-abhishekpal provided.
Is also a fresh install

Version: v0.20.3
Model: PI4 B

nicomiguelino, 2025-10-01:
@TRS4242, @me-abhishekpal, @DerCookieLiker 
The fix is on the way. I just merged the fix.

nicomiguelino, 2025-10-01:
@TRS4242, @me-abhishekpal, @DerCookieLiker
I updated Anthias on my Pi 4 and 5 and it works fine now.

Please let me know if you're still having issues.

closed 2025-10-01 by nicomiguelino

TRS4242, 2025-10-01:
@nicomiguelino Thank you! That seemed to fix my issue!

me-abhishekpal, 2025-10-07:
Thank you, I am yet to test and will report back.

PR #2521 (2025-10-01): fix: downgrade `sh` from 2.2.2 to 1.8
### Issues Fixed

Fixes #2520

### Description

Merging the PR that upgrades `sh` from 1.8 to 2.2.2 broke the viewer. More details can be found at #2516.

### Checklist

- [x] I have performed a self-review of my own code.
- [x] New and existing unit tests pass locally and on CI with my changes.
- [x] I have done an end-to-end test for Raspberry Pi devices.
- [x] I have tested my changes for x86 devices.
- [x] I added a documentation for the changes I have made (when necessary).